kliment/Sprinter #218: dual-license file fastio.h: an alternative license

DerAndere1, 2021-02-24:
I would like to use the content of file fastio.h for teaching, but I do not like to teach methods that are only usable in GPL licensed code. Are you still in contact with Triffid_Hunter and could you allow me to redistribute the file fastio.h (and **only that one file**) under the [Apache-2.0](https://directory.fsf.org/wiki/License:Apache-2.0) license? The Apache-2.0 license would allow reuse of that code in closed-source software but still requires attribution in all forms of distribution (binary or source code) . I know that this topic is highly debatend amongst the free/open source community, so thanks for your consideration.

kliment, 2021-02-24:
@triffid what do you say?

DerAndere1, 2021-02-24:
I can think of the following procedure. If you have suggestions, I am open for feedback:
1)  I plan to add a NOTICE file with the following content:
```
This work contains the software fastio (fastio.h)
by Triffid_Hunter and Kliment.

fastio (https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h)
Copyright 2011 Kliment
Copyright 2011 Triffid_Hunter
This code contibuted by Triffid_Hunter and modified by Kliment
```

2) I plan to add a LICENSE file of my repository that contains the following text: 
```
===============================================================================
For fastio:

fastio.h

fastio (https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h) is 
Copyright 2011 Kliment
Copyright 2011 Triffid_Hunter 
Licensed under the GNU GPL v3 or (at the user's discretion) any later version OR (at the user's discretion) licensed under 
the Apache License, Version 2.0.
For details, see file LICENSES/GPL-3.0.txt or file LICENSES/Apache-2.0.txt, respectively.

===============================================================================
```

3) I plan to copy the contents of file fastio.h from the last commit that has only the two of you as contributors ( https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h )

4) I plan to later add the following lines to my copy of fastio.h without removing the existing attribution statement:
```
/* SPDX-License-Identifier: GPL-3.0-or-later OR Apache-2.0 
 * 
 * uno_fastio
 * Copyright 2021 ITlernpfad
 * 
 * This work contains the software fastio (fastio.h) 
 * by Triffid_Hunter and Kliment, with modifications by ITlernpfad
 * 
 * fastio (https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h)
 * Copyright 2011 Kliment
 * Copyright 2011 Triffid_Hunter
 * /
/*
	This code contibuted by Triffid_Hunter and modified by Kliment
	why double up on these macros? see http://gcc.gnu.org/onlinedocs/cpp/Stringification.html
*/
```

triffid, 2021-02-25:
How about LGPL?

That allows linking into proprietary/closed code without forcing that code to be opened (which you want) while retaining the requirement that if you improve fastio itself, you must share (which we want).

LGPL has a lot of verbiage around static/dynamic linking, but I think it should apply similarly to header macros as well..?

DerAndere1, 2021-02-25:
yes, LGPL is also what the official Arduino core uses and the terms of LGPL are compatible with my usecase. I read [somewhere](https://www.quora.com/What-are-the-major-differences-between-GNU-LGPL-v3-and-v2-1?share=1) that LGPL-2.1-licensed code cannot contain code licensed under Apache-2.0 [or 2-clause-BSD](https://softwareengineering.stackexchange.com/questions/189668/lgpl-2-1-vs-lgpl-3-0-advantages-and-disadvantages). Thus, to allow future contributions of Apache-2.0-licensed code, I would opt for LGPL-3.0-or-later. If you agree, please write an me explicit permission here or elswere, something like 
"I agree that the file fastio.h  from https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h  is dual-licensed under the GNU GPL v3 or (at the user's discretion) any later version OR under the terms of the LGPL-3.0-or-later (https://spdx.org/licenses/LGPL-3.0-or-later) license. " 

Once both of you wrote that permission, I could then release a fastio library under the LGPL-3.0-or-later.

kliment, 2021-02-25:
I agree that the file fastio.h from https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h may be dual-licensed under the GNU GPL v3 or (at the user's discretion) any later version OR under the terms of the LGPL-3.0-or-later (https://spdx.org/licenses/LGPL-3.0-or-later) license. :)

triffid, 2021-02-25:
Whereas I am the original author and copyright holder of the contents of https://github.com/triffid/ATmega-Skeleton/blob/496d58f357d9509550a77e0696f9b77f8feb7e9e/mendel/arduino.h, and

Whereas https://github.com/triffid/ATmega-Skeleton/blob/496d58f357d9509550a77e0696f9b77f8feb7e9e/mendel/arduino.h predates the derivative work posted at https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h by ~18 months ± a few days, and

Whereas all subsequent author(s) of the derivative work https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h (ie @kliment) agree to the licensing change discussed above,

I declare that https://github.com/triffid/ATmega-Skeleton/blob/496d58f357d9509550a77e0696f9b77f8feb7e9e/mendel/arduino.h (and by extension https://github.com/kliment/Sprinter/blob/b7b099bb0e0a8797b71abdf50e3da6880a86d967/Sprinter/fastio.h) is hereby available to be used under the terms of the LGPL-3.0-or-later (https://spdx.org/licenses/LGPL-3.0-or-later) license.

--------

Thank you for caring about open source licensing and opening this discussion rather than simply assuming that the code is public domain!

 We should have been more sensible about this from the start I guess..

DerAndere1, 2021-02-25:
Much appreciated. I will close this constructive discussion as resolved. The new library can be found under the name uno_fastio at gitlab.com. Link: https://gitlab.com/itlernpfad_public/uno_fastio

closed 2021-02-25 by DerAndere1

triffid, 2021-02-25:
Curious, howcome you wanted to license this file, but then don't use the READ/WRITE/etc macros in your examples?

kliment, 2021-02-25:
I strongly suspect the examples can change now that that file is there :)

DerAndere1, 2021-02-25:
Yes exactly. The example was written live in class and pupils had to be introduced to macros on the fly. So I teach them one layer of abstraction after the other.  ;)

triffid, 2021-02-26:
Fun, so you're gonna boggle their brains with these string-wrangling macros next?